echo-lalia/MicroHydra #2: Attempting to burn your "MicroHydra" upload on M5Burner bricks the unit

Vaygrim, 2024-02-06:
Brand new, just got it today. Burning the firmware image (v0.2) that you uploaded to M5Burner results in my unit being bricked. After burning finished, instead of rebooting the screen just went black and now will no longer boot. No error messages, nada. Just a black screen.

Attempted to burn the original core image back to it, but M5Burner won't write to it, says it can't connect over the ComPort.

echo-lalia, 2024-02-07:
Did you erase the flash before burning the firmware?

Either way, you can put the device into bootloader mode by making sure the power on the device is off, then holding G0 as you plug it into a computer. From there, you can burn whatever firmware you want on it :)

Vaygrim, 2024-02-07:
"Holding the G0 while you plug it in" WOAH, I didn't even know this was an OPTION!
Thank you for the info! That isn't indicated in M5Burner where I tried imaging it, so I figured it was a bad upload or something.

closed 2024-02-07 by Vaygrim

echo-lalia, 2024-02-07:
Yeah I with they made it more clear, as a lot of people seem to be having the same issue as you. 

As far as I know, (unless the firmware was intentionally designed to break stuff), you cant brick the device just by uploading a bad firmware to it :) You can always force it to go back into bootloader mode and put something new onto it.

Vaygrim, 2024-02-07:
By the way, did you know about the 'secret button' under the "M5Stamp S3" sticker on top?
Middle of the sticker, there is a black oval with "BTN" written in text along with a smaller red oval on it. You can use the included hex wrench to press that down while plugging the unit in to USB. Sets it into some kind of "Downloader Mode". That's how I fixed my issue.