echo-lalia/MicroHydra #3: slow st7789 driver

Lana-chan, 2024-02-26:
is there any reason why you're not using the compiled mpy version? (other than it's extra steps)

https://github.com/devbis/st7789_mpy

echo-lalia, 2024-02-26:
Great question! 

I'm assuming you're talking about the version that's rebuilt as a C module, but if you're talking about another one, hopefully this still answers your question.

This is a question that has been discussed on the Cardputer discord, and I went back and forth on how I felt about it, but ultimately I decided to avoid using a compiled display driver to increase MicroHydras accessibility. 

One of my main goals for MicroHydra was to make it as accessable as possible for new developers who want to learn to tinker and create scripts for their cardputer. I worry that by switching to a specific firmware, I'll be increasing the barriers-to-entry for people who might have otherwise wanted to build an app for MicroHydra. 

After some experimentation, it seems like the main difference in speed between the .py driver, and the compiled C driver, is that the compiled version uses a framebuffer (whereas the .py version is writing every change directly to the display). 

I modified the st7789py into a driver I called st7789fbuf, which is basically just the same driver, but reworked to make use of the fast methods provided by the built-in FrameBuffer class as much as possible, and from some very non-scientific testing, the performance seems to be nearly the same as the C version. 

I'm also experimenting with some changes that will allow the driver to use multiple framebuffers for different regions of the screen, which would allow things like the status bar to be updated separately from the scrolling elements. Once these changes are finished, I'm hoping to move the launcher over to the new driver.

I think it's also worth pointing out, that I'm providing compiled .mpy versions of all the built-in files in the releases section, and a fully frozen/compiled version of MicroHydra over in the 'microhydra-frozen' repository, both of which provide faster load times for the launcher. But, the regular .py files are nice too, I think, because you should just be able to use them with any version of MicroPython 😁

Lana-chan, 2024-02-26:
i didn't know there was a discord 👀

i've uploaded the prebuild binary in https://github.com/russhughes/st7789_mpy to my cardputer and i've been trying that instead of st7789py. it doesn't seem to be framebuffer based at all, the text writing happens just the same, but it takes a lot less time to write a whole screen's worth of text. i wonder if we're talking about the same thing?

echo-lalia, 2024-02-26:
Yeah! It's not an official discord, just something that was linked on Reddit, but there's some cool people in there who I've been enjoying talking to about this project :)

And i think you're right. I was combining S3LCD and st7789_mpy in my head. I do still think most of what I said is true either way though.

If you haven't checked it out already, I made a simple 'flappy bird'-like game, using only primitives provided by the st7789fbuf driver, over on the [apps](https://github.com/echo-lalia/MicroHydra-Apps/tree/main/apps) repository :)

I'm sure there's still speed improvements that can be made, but if you were messing with the st7789py version before, you might be surprised by how much faster the fbuf version is.

echo-lalia, 2024-02-26:
I'd like to add a little clarification though, that I'm fully in support of people using whatever display drivers they'd like for apps they build for MH!  It's just for the "base" launcher that I feel it would be better to keep it as open as possible.

Lana-chan, 2024-02-26:
i pinged you on that discord about this already but just for the sake of this thread, i've tried the fbuf driver and my irc client seems to draw text better (even if just using the bitmap text call instead of the "quick" text one). thanks!

closed 2024-02-26 by Lana-chan

Lana-chan, 2024-02-26:
also let it be known that i completely neglected this repo's release zip containing .mpy files, which is what i was looking for in the first place. oops