echo-lalia/MicroHydra #27: SD card is not showing up in UI

puffballbird, 2024-03-16:
I have seen no sd card or and flash pop up 
I dont know is im just not doing something or im missing something
I flashed it on to my cardputer from the 5m burner and made a folder for the py files and notes
But It does not show up anywhere 
Some help would be nice thank you

 oh and if you dont mine a seggestion 
people showed a software similar to this booting and launching bin files or applications 
Maybe the capabillity to boot up bins like other launchers would expand the capbilties of the laucher
:) also its cools stuff what yall are doing

echo-lalia, 2024-03-16:
Hi! You need to use a tool such as Thonny in order to view the files on the device and copy things over. 

Take a look at the installation tutorial on the homepage of this Github. You can skip the parts about flashing micropython (because you already flashed it with M5 burner) and you can skip copying the microhydra files over too. 

The first time MH starts up, it creates an apps folder on the main flash, and if it can read your sd card, it'll make an apps folder there too :)

puffballbird, 2024-03-16:
Problem its seems like its not automatically making a apps folder on the sd Im in the prosses of figuring out why Thonny does not have the option for esp32-s3 is there like a id or link or something i need to download for it to recognize it

echo-lalia, 2024-03-16:
Does the microhydra launcher start up when you power on the device? If so, you can skip all of those steps. Just look at the part about transferring files to the device! 

And for the SDcard, it must be formatted as fat32 in order to work.

puffballbird, 2024-03-16:
oh i just used ruffus to format it to fat32 and it it poping up in files
but im going to look into file transfer for the on board flash thank you 
for the help :)

closed 2024-03-17 by echo-lalia