echo-lalia/MicroHydra #31: Is it possible to add a micropython repl environment to enhance functionality?

maysrp, 2024-03-24:


maysrp, 2024-03-24:
I tried to use [fbconsole](https://github.com/boochow/FBConsole) to build the micropython repl environment, but it failed.

echo-lalia, 2024-03-24:
Someone has actually built a little [repl-like app](https://github.com/AskewedBox/Cardputer-MicroHydra/tree/main/MicroHydra/apps) for MicroHydra, and I think it's pretty shnazzy 😁

I think it might be possible to actually connect to the full micropython repl from an app. It's something I'd like to try eventually, but I'm not sure when I'll be able to get to it.

TheOddCell, 2024-08-26:
There is a terminal that echo-lalia (the creator of MicroHydra) has made, however I don't know if it is really a REPL:
PR #111: Create simpleterminal

echo-lalia, 2024-08-26:
> There is a terminal that echo-lalia (the creator of MicroHydra) has made, however I don't know if it is really a REPL: PR #111: Create simpleterminal

I'm happy you're interested in my new changes! Unfortunately, that's not an actual 'terminal' (maybe I should've picked another name? I couldn't think of anything more descriptive, haha). 

All that module does is add a lightweight terminal-like *graphics* object, which apps can use to print status information to the REPL and to the display simultaneously. Currently it's only used by the `launcher/getapps.py` built-in app.

<br/>

<br/>

That said, RealClearwave has also added an actual built-in [terminal](https://github.com/echo-lalia/MicroHydra/commit/207a608f62420fd4dc2c61d485282713a26f65d5)! It's not a MicroPython REPL however, so it doesn't quite complete this feature request.

RealClearwave, 2024-08-30:
What about dumping Esp32-s3's flash directly using esptool.py?

echo-lalia, 2024-10-01:
Hello! The Terminal app has been extended to include REPL functionality (In addition to its linux-style commands). This completes this feature request 😁

closed 2024-10-01 by echo-lalia