echo-lalia/MicroHydra #33: Adding support for Daylight Saving Time Adjustment

swagatpanda, 2024-03-25:
Hi , 

Amazing library, made my entry into the world of MicroPython so easy.

Can I tinker with the code to add an option in the settings menu to enable Daylight Saving Time ?

You can let me know if that is the right way to do it or you have something else in mind

echo-lalia, 2024-03-25:
Feel free to tinker with the code however you'd like. If you've added a feature, or improved it in some way, and want to share, submit a pull request, and I'll merge it into the project (unless it breaks something else, of course) 😊

The current timezone logic is extremely 'naive'; it literally just adds or subtracts some number from the current hour. Adding support for daylight savings time is gonna require expanding that logic somewhat. 

Also! I'm in the middle of completely rewriting the settings app with the help of a new module called "HydraMenu". The old settings app code straight-up sucks and it's really annoying to edit. Please don't waste your time trying to tinker with that. The new settings app is in the "settings-overhaul" branch, and it should be very easy to add a new settings option there 😁

swagatpanda, 2024-03-25:
I'll check the settings-overhaul branch in that cases 👍

closed 2024-03-25 by swagatpanda