echo-lalia/MicroHydra #37: End of lines getting stripped away in the editor

h-david-a, 2024-04-14:
One-two symbols in the line ends getting stripped when opening files in editor.
Seems like the issue is with the code handling line breaks (\r\n) and trimming 2 last chars afterwards, and repeats once again:
#string formatter
def remove_line_breaks(line):
    if line.endswith('\r') or line.endswith('\n'):
        line = line[:-2]
    if line.endswith('\r') or line.endswith('\n'):
        line = line[:-2]
    return line

I think, only last char shoud be trimmed:   line = line[:-1]

h-david-a, 2024-04-14:
Can't assign this issue to myself. Will just fork and PR back

h-david-a, 2024-04-14:
https://github.com/echo-lalia/Cardputer-MicroHydra/pull/38

echo-lalia, 2024-04-14:
Great point. Thanks for sharing this issue!

closed 2024-04-14 by echo-lalia