echo-lalia/MicroHydra #47: release v1. "copy" file function not working

h-david-a, 2024-04-16:
'Copy' Code in files.py is commented out.
Is there a reason for that?

echo-lalia, 2024-04-16:
Darn, that's an oversight. Thanks for pointing it out!

The previous version just duplicated/renamed the files in the same location and didn't offer a way to move them. I changed the implementation to allow you to copy files between folders, and I removed the renaming popup to make the app behave more similarly to file browsers on other operating systems. 

I'll have to add a workaround for this in the next version :)

h-david-a, 2024-04-16:
Fixed with adding .bak extension.
Messing with sequential numbering seemed a bit overkill. And adding random sequences creates mess.
Most obvious scenario where one copy-pastes file in the same place would be a backup. Hence .bak )