echo-lalia/MicroHydra #48: Adding items in settings.py crashes settings app

h-david-a, 2024-04-16:
I was trying to add 24h clock setting to settings.py (there were several iterations, ended up with this):
```
menu_def = [
    (HydraMenu.IntItem, 'volume', {'min_int':0,'max_int':10, 'instant_callback':update_config}),
    (HydraMenu.RGBItem, 'ui_color', {'instant_callback':update_config}),
    (HydraMenu.RGBItem, 'bg_color', {'instant_callback':update_config}),
    (HydraMenu.WriteItem, 'wifi_ssid', {}),
    (HydraMenu.WriteItem, 'wifi_pass', {'hide':True}),
    (HydraMenu.BoolItem, 'sync_clock', {}),
    **(HydraMenu.BoolItem, 'clock_24h', {}),**
    (HydraMenu.IntItem, 'timezone', {'min_int':-13,'max_int':13}),
    ]
```

Launching settings.py causes black screen.

'log.txt': 
```
Tried to launch '/launcher/settings.py', but failed: '24h_clock'
Tried to launch '', but failed: ''
Tried to launch '', but failed: ''
Tried to launch '/launcher/settings.py', but failed: 'name 'false' isn't defined'
Tried to launch '', but failed: ''
Tried to launch '', but failed: ''
Tried to launch '/launcher/settings.py', but failed: 'clock_24h'
Tried to launch '', but failed: ''
Tried to launch '', but failed: ''
Tried to launch '', but failed: ''
Tried to launch '', but failed: ''
Tried to launch '/launcher/settings.py', but failed: 'clock_h'
```

echo-lalia, 2024-04-16:
I think that's a KeyError. 

The settings app uses the [mhconfig](https://github.com/echo-lalia/Cardputer-MicroHydra/blob/main/MicroHydra/lib/mhconfig.py) module to create and manage user settings in config.json. 

The config object works like a dictionary, and the settings app expects the settings names to exist as keys in the config.   
The config doesn't have a value for 'clock_24h' so it errors when it tries to read the default/previous value.

echo-lalia, 2024-04-16:
That's a good question.

The current implementation is just using the built-in ntptime module, along with simple addition, to set the time and apply the timezone. 

It works like that because it was the simplest way for me to implement it when first creating the launcher, not because I think it's a good way to do it 😅 It might be a good idea to try a different approach.

</br>

If you have an interest, feel free to try making that change and see how it goes, but I'll warn you that the launcher is super sensitive to memory issues due to the framebuffer/SDCard/audio/WiFi/etc all being used simultaneously, and so sometimes really random things can cause instability, so this change would definitely need to be tested before merging it in.   
This change could just as easily use less memory than the old implementation though, so I'll be curious to see :)

closed 2024-04-18 by echo-lalia