echo-lalia/MicroHydra #50: We need popup_options_2d

ndrnmnk, 2024-04-17:
I already made ChatGPT do coding, but it looks ugly.
You can insert this into lib/mhoverlay.py to see how it works:

    def popup_options_2d(self, options, title=None, shadow=True, extended_border=False, scrollable=False):
        """
        Display a popup message with given options arranged in 2D grid.
        Blocks until option selected, returns selection.
        """
        tft = self.display
        num_cols = 5  # Number of columns in the grid
        per_colomn = 6
        col_width = 240 // num_cols  # Width of each column
        row_height = 16  # Height of each row
        
        if scrollable:
            old_options = options
            options = old_options[:num_cols*per_colomn]

        if self.compatibility_mode:
            # draw box
            box_height = (len(options) // num_cols + 1) * row_height + 8
            box_y = 67 - (box_height // 2)
            if extended_border:
                tft.fill_rect(0, max(box_y - 10, 0), 240, min(box_height + 20, 135), self.config.palette[1])
            if title:
                self.draw_textbox(title, 120, box_y - 16, shadow=shadow, extended_border=extended_border)
        else:
            # draw box
            box_height = (len(options) // num_cols + 1) * row_height + 8
            box_y = 67 - (box_height // 2)
            if extended_border:
                tft.fill_rect(0, max(box_y - 10, 0), 240, min(box_height + 20, 135), self.config.palette[1])
            if title:
                self.draw_textbox(title, 120, box_y - 14, shadow=shadow, extended_border=extended_border)

        cursor_index = 0
        keys = self.kb.get_new_keys()
        while True:
            if self.compatibility_mode:
                # draw options
                for idx, option in enumerate(options):
                    col = idx % num_cols
                    row = idx // num_cols
                    x = col * col_width + (col_width - len(option) * 4) // 2
                    y = box_y + row * row_height + 4
                    if idx == cursor_index:
                        tft.fill_rect(x - 2, y - 2, len(option) * 4 + 4, 20, self.config.palette[0])
                        tft.text(self.font, option, x, y, self.config.palette[5], self.config.palette[0])
                    else:
                        tft.text(self.font, option, x, y, self.config.palette[4], self.config.palette[2])
            else:
                # draw options
                for idx, option in enumerate(options):
                    col = idx % num_cols
                    row = idx // num_cols
                    x = col * col_width + (col_width - len(option) * 4) // 2
                    y = box_y + row * row_height + 4
                    if idx == cursor_index:
                        tft.fill_rect(x - 2, y - 2, len(option) * 4 + 4, 20, self.config.palette[0])
                        tft.text(option, x, y, self.config.palette[5])
                    else:
                        tft.text(option, x, y, self.config.palette[4])
                tft.show()

            keys = self.kb.get_new_keys()
            for key in keys:
                if key == ";":  # Move cursor up
                    cursor_index = (cursor_index - num_cols) % len(options)
                elif key == ".":  # Move cursor down
                    cursor_index = (cursor_index + num_cols) % len(options)
                elif key == ",":  # Move cursor left
                    cursor_index = (cursor_index - 1) % len(options)
                elif key == "/":  # Move cursor right
                    cursor_index = (cursor_index + 1) % len(options)
                elif key == "w" and scrollable:
                    next_page_result = self.popup_options_2d(options=old_options[num_cols*per_colomn:], title=title, shadow=shadow,
                                                          extended_border=extended_border, scrollable=True)
                    if next_page_result != None:
                        return next_page_result
                    else:
                        cursor_index = 0
                elif key == "`" or key == "ESC" or key == "BSPC" or key == "a":
                    return None
                elif key == "ENT" or key == "GO":
                    return options[cursor_index]

            time.sleep_ms(10)

echo-lalia, 2024-04-17:
Just edited for readability :)

I think that sounds like a sensible addition, and it could make it easier to make a nice looking UI on some apps.   

I like ChatGPT for some stuff, but I don't really trust its code for something like this, just because usually it seems to create code that's a little hard to interpret and edit, so I'd probably just be rewriting it all to ensure I know what it's doing. The mhoverlay module could use some beautifying, anyways.

echo-lalia, 2024-08-07:
FYI I have completely rewritten the lib.mhoverlay module (now called lib.hydra.popup) for what I intend to be MicroHydra 2.0, and I kept your suggestions in mind when I did it.

The new popup_options method can create both 1d and 2d options (and gives you control over the arrangement of the columns; columns can be different lengths and are arranged nicely). 

The API is a little different than what you described here (and I have not implemented scrolling functionality in it yet), but I think it otherwise has the functionality you were looking for :)

If you want to check it out, the changes have been pushed to the experimental-multiplatform branch! But, it isn't compatible with MicroHydra 1.0, and it might still be changed more before I'm done.

ndrnmnk, 2024-08-09:
That's exactly what i wanted, thanks a lot! I'll port IR app to work with that branch in a few hours

closed 2024-08-09 by ndrnmnk

echo-lalia, 2024-08-09:
There are quite a few changes in the experimental branch, and more will probably be coming soon. I also haven't written any docs for the new modules yet, so no need to rush on those updates :) I'm really glad you're exited about the changes, though!

ndrnmnk, 2024-08-10:
It's too late, already ported the app and it works with these changes too