echo-lalia/MicroHydra #56: Question, lib packages where

puffballbird, 2024-05-03:
Hay is there a place with the full package of all the micropython libs used in apps or micro python used in microhydra 
so i can plug the lib Package into Thonny so I can start to write up some apps and code
it would be nice to know so i can start to figure stuff out about how to make apps 

also is there any network examples i can look at
I'm hopping to make a app to allow the carputer to use a ESB-NOW chat like system and some other stuff

sorry if i am just blind and couldn't find it
please correct me on anything I'm exited to learn and work on this
thank you for your time :)

echo-lalia, 2024-05-03:
MicroHydra specific libs are all [here](https://github.com/echo-lalia/Cardputer-MicroHydra/tree/main/MicroHydra/lib). 

For other MicroPython modules, you can look [here](https://github.com/micropython/micropython-lib/tree/master) or in the main MicroPython repo, but many of the built-in MicroPython modules are implemented in C.

puffballbird, 2024-05-03:
thank you very much

closed 2024-05-03 by puffballbird