echo-lalia/MicroHydra #58: Running micropython apps

Mattjcuk, 2024-05-25:
Hi,

I have a basic app which I have written using uiFlow 2 for the cardputer(using the envro IV sensor), the app runs fine on the cardputer and I can upload it to the device and it will run fine when I unplug it from my computer.  I downloaded the main.py from the cardputer (this is my sensor code), renamed it and then uploaded it to the apps folder on my sd card for microhydra, microhydra sees it but I get an error message saying the M5 lib is unavailable, however it is on the device somewhere as the app will run separately from microhydra .  I have tried one of the example programs and this works fine.  So everything works independantly but when launching the app from Microhyrda it doesn't.

Do you have any ideas on this please?

Many thanks,

Matthew

echo-lalia, 2024-05-26:
Hey! 

Thanks for reaching out, and I'm sorry for the confusion you had while trying to develop with MicroHydra.

Unfortunately, this isn't really a glitch, or an unexpected behaviour; the M5 modules used in UIFlow aren't currently available in MicroHydra, so any code that uses those libs would need to be adapted to use standard MicroPython modules (or the modules that come with MicroHydra, or any other portable modules you make/find).

The M5Unified library (and any other module from UIFlow with M5 in the name) are all written in C and compiled alongside MicroPython for the UIFlow firmware. I have an interest in potentially adding those modules in MicroHydra's firmware, too, but I found it kinda difficult to separate the M5 libraries from the rest of UIFlow, and I put that on hold while I work on other aspects of MicroHydra. 

<br/> 

If you're determined to make your app work on MicroHydra, I might be able to point you in the right direction. The [store page](https://shop.m5stack.com/products/env-iv-unit-with-temperature-humidity-air-pressure-sensor-sht40-bmp280) for your sensor notes that the device uses I2C to communicate, along with the addresses used:    
`I2C communication interface (SHT40 I2C address: 0x44, BMP280 I2C address: 0x76)`   
The built in [MicroPython I2C](https://docs.micropython.org/en/latest/library/machine.I2C.html) module should be able to fetch that sensor data. I don't have your sensor so I can't test this myself, but I imagine simply defining the correct I2C Pins for the CardPuter, and then running a `I2C.readfrom(addr, nbytes)` would get the sensor data. 

Whatever you end up doing; good luck with your app! Feel free to reach out with a follow up question if you get stuck again :)

Mattjcuk, 2024-05-26:
Hi,

Many thanks for the detailed explanation and solution, I will have a look at your suggestion.

Thank you again,

Kind regards,

Matthew

closed 2024-08-18 by echo-lalia