echo-lalia/MicroHydra #77: BUG: (for frozen version): Does not install correctly when used with M5launcher

TheOddCell, 2024-08-20:
What the title said. It does not work correctly when used with M5launcher, and this is a problem for me because I use m5launcher a lot to switch firmware.

echo-lalia, 2024-08-21:
Could you elaborate a bit on what you mean? In what way does it not work, and what version of both are you using?

MicroHydra (along with every MicroPython-based firmware) didn't work with M5Launcher for a long time, but the M5Launcher creator has changed things on their end to allow it to work. As far as I know things should still be working the same way, but I could be wrong.

TheOddCell, 2024-08-22:
It would sometimes bootloop (the m5launcher screen would keep appearing), or just have a black screen.

TheOddCell, 2024-08-23:
I have found a way to get the frozen version to work. You just need to copy the ENTIRE SOURCE of the same version and put it on the root of the SD card inserted to the CardPuter. Closed!

closed 2024-08-23 by TheOddCell