echo-lalia/MicroHydra #113: BUG: but failed: 'no module named 'xxx''

TheOddCell, 2024-08-26:
I am trying some apps, and they all have similar problems.
Tried to launch '/sd/apps/chaosDice.py', but failed: 'no module named 'lib.display''
Tried to launch '/sd/apps/FancyClock.py', but failed: 'no module named 'lib.display''
(Both from Microhydra-Apps)
MicroHydra v1.23.0 on esp32 (cardputer), with MicroPython being 1.23.0.
Ran directly on firmware (no M5launcher or anything, not frozen)

echo-lalia, 2024-08-26:
Hi! Those errors indicate that you're using the previous (stable) version of MicroHydra. To get apps for your version, see [here](https://github.com/echo-lalia/MicroHydra-Apps/tree/v1.x).

Some built-in modules have been moved/renamed/modified for MicroHydra v2.0, to allow for more future growth (like supporting new devices), and so the newest versions of most apps wont work with the previous version of MicroHydra.  
If you wanna check out the newest changes, take a look at this preview release [here](https://github.com/echo-lalia/MicroHydra/releases/tag/v2.0-preview) (You'll need to download one of the Cardputer-specific zip files from that release). But, I'd like to officially release v2.0 soon 😁

TheOddCell, 2024-08-27:
Thank you for your answer! Case closed!