echo-lalia/MicroHydra #143: Doesn't start on Cardputer

itJunky, 2024-09-05:
I'm using a M5Launcher for starting any .bin files.

I'm copied CARDPUTER.bin to SD, renamed it to MicroHydra, unzip and copy  CARDPUTER_compiled.zip to root dir on flash, and inserted SD to cardputer. After boot m5launcher i select MicroHydra,bin, press install and waitng a progressbar. 
But after it goes to end everything is stuck (((

sanketkorde, 2024-09-05:
Give this a shot
https://bit.ly/4eaAnWN
Passcode: changeme
you must have a gcc compiler

echo-lalia, 2024-09-05:
Hello!

Most of my testing is done on the Cardputer, so I'd definitely expect it to work, but I'm not sure why you're having issues. 

If you put it into flashing mode by plugging it in, holding G0, and hitting the reset button, are you able to flash it successfully? 

Finally, which .bin did you download; the v2.0 bin, or the 2.1 preview?

echo-lalia, 2024-09-05:
*btw: you don't need to download the `compiled.zip` if you're flashing the firmware :) that zip is just for installing on top of normal MicroPython. The MicroHydra firmware .bin has those files frozen into it.*

closed 2024-09-24 by echo-lalia