echo-lalia/MicroHydra #184: Improving upon and documenting MountSD

Avginike, 2025-02-02:
Hello! 
I think that the idea behind MountSD is wonderful.
However, it is lacking documentation and doesn't currently seem to work?
What does it do exactly and how does one use it?

echo-lalia, 2025-02-04:
That script is super simple, and is only responsible for importing and running the code from the new MicroHydra SDCard module.

How have you tested it? (What were you expecting to happen, and what happened instead?)

Avginike, 2025-02-04:
"Use this simple tool to mount your SD card. This is useful for transferring files via USB and for editing apps on the SD card."

echo-lalia, 2025-02-11:
No worries. The script mounts the SDCard to the MicroPython filesystem, so that you can access it from Thonny (or any other similar tool).

It won't mount to your PC as a storage device or something like that, unfortunately, though.