echo-lalia/MicroHydra #191: Add support for ESPBOY

coppaxamsukk, 2025-03-12:
Espboy is a diy esp8266 based project that's similar to cardputer but available from long ago. 

Can you port the awesome project for espboy?
https://www.espboy.com/

[image]

echo-lalia, 2025-03-13:
Sounds like a cool project!

Unfortunately I don't have one of these, so I can't personally do the port.  
But, I'll leave the issue open in case anyone else wants to take a crack at it (and feel free to message me for help, if you do!)

coppaxamsukk, 2025-03-13:
You don't need espboy, just esp8266 can do.
And it's full diy.

altgreen64, 2025-03-13:
Ladies and Gentlemen! The dream device has been created
https://www.clockworkpi.com/picocalc
I can't wait to start creating software for it

coppaxamsukk, 2025-03-13:
> Ladies and Gentlemen! The dream device has been created
> https://www.clockworkpi.com/picocalc
> I can't wait to start creating software for it
> 
> czw., 13 mar 2025 o 06:35 coppaxamsukk ***@***.***>
> napisał(a):
> […](#)

You think that can surpass cardputer? I just wanna know. It's looking awesome but cardputer uses esp32 that's a lot powerful than rpi pico /stm32.

coppaxamsukk, 2025-03-13:
And espboy already has a lot of features like GSM phone
GPS navigator
Meteostation
LORA messenger
RFID/NFC reader/repeater
Sub1Ghz inspector
Sub1Ghz scanner
MOD/S3M/XM/IT player
AY38910/12 Yamaha chip player
Web radio
FM radio
Walkie-talkie
Meccano rover
Flower watering
Ice fishing automation
Volt/Amper meter
CW radio beacon
RSS reader
Digital ruler
Pinger
Digital thermometer
Heart rate meter
WiFi analizer
WiFi sniffer
WiFi packet monitor
WiFi pinger
WiFi deauther
IR blaster
Geiger counter
I2C scanner
MP3 player

echo-lalia, 2025-03-13:
> You don't need espboy, just esp8266 can do.
> And it's full diy.

MicroPython already supports esp8266! 

If I wanted to port MicroHydra's libraries to any new device, i need the entire device, haha. For example, I couldn't really port the display driver without the new display... And the same goes for all the other peripherals, unfortunately.

coppaxamsukk, 2025-03-13:
> > You think that can surpass cardputer? I just wanna know. It's looking awesome but cardputer uses esp32 that's a lot powerful than rpi pico /stm32.
> 
> Honestly I really love my rp2040 boards. The raw performance is a bit lower, but the 8 programmable state-machines (and lots of readily available MicroPython modules that utilize them) makes them really capable little devices.
> 
> The picocalc seems to also support the pico2, which would be faster! (I couldn't find any benchmarks comparing it to an esp32s3, but [here's](https://icircuit.net/benchmarking-raspberry-pi-pico-2/3983) one that compares it to an esp32 in integer and floating point performance!)

Yeah i like it too, i have one. It's small in size and also has built in usb storage support, better for micro/circuit Python, that isn't available in esp32 devboard.